jiegec/usbip #2: Prevent panic on invalid interface number

linkmauve, 2021-04-26:
At https://github.com/jiegec/usbip/blob/master/src/device.rs#L349, `setup_packet.index` is sometimes (much) larger than the number of interfaces, I got 514, 1284, 264, 520, 1034 and 1290 for instance on a device with only 12 interfaces.

Since all of them seem to be in range modulo 256, I could add such a modulo but I’d prefer someone accustomed to USB to verify that this is indeed the correct fix.

The modulo otherwise fixes this panic:
```rust
thread 'tokio-runtime-worker' panicked at 'index out of bounds: the len is 12 but the index is 514', src/device.rs:349:37
```

jiegec, 2021-04-27:
Oh, that's weird.

jiegec, 2021-04-27:
Yes, you are right:

[image]

linkmauve, 2021-04-27:
This indeed fixes the issue, thanks!

closed 2021-04-27 by linkmauve