jiegec/usbip #4: The Direction enum seems to be shadowed

voidfield101, 2022-03-16:
The Direction enum seems to be shadowed by the rusb crate.

Importing Direction will result in the following error:
```
error[E0603]: enum `Direction` is private
 --> src/cdc.rs:8:94
  |
8 | use usbip::{UsbEndpoint, EndpointAttributes, UsbInterfaceHandler, UsbInterface, SetupPacket, Direction};
  |                                                                                              ^^^^^^^^^ private enum
  |
note: the enum `Direction` is defined here
 --> /home/void/.cargo/registry/src/github.com-1ecc6299db9ec823/usbip-0.4.0/src/lib.rs:6:5
  |
6 | use rusb::*;
  |     ^^^^^^^
```

At the same time the consts crate is also not marked as public making a direct import with ``use usbip::consts::Direction`` impossible.

jiegec, 2022-03-17:
Nice catch, maybe we should re-export Direction from rusb instead.

jiegec, 2022-03-17:
See commit 7b0f47c78bf32137b24a1ee826af5af40e05301d

closed 2022-04-11 by jiegec