jiegec/usbip #5: Cannot use library because of error E0658

Clemens-Dautermann, 2022-07-14:
If I include the library in my Cargo.toml the project will not compile because of the following error:
```
error[E0658]: use of unstable library feature 'thread_local_const_init'
  --> /home/******/.cargo/registry/src/github.com-1ecc6299db9ec823/tokio-1.20.0/src/coop.rs:34:1
   |
34 | / thread_local! {
35 | |     static CURRENT: Cell<Budget> = const { Cell::new(Budget::unconstrained()) };
36 | | }
   | |_^
   |
   = note: see issue #84223 <https://github.com/rust-lang/rust/issues/84223> for more information
   = help: add `#![feature(thread_local_const_init)]` to the crate attributes to enable
   = note: this error originates in the macro `$crate::__thread_local_inner` (in Nightly builds, run with -Z macro-backtrace for more info)
```
It seems to have something to do with tokio that is used by usbip.

jiegec, 2022-07-15:
The usbip crate depends on tokio 0.17.0:

```toml
[dependencies]
tokio = { version = "1.17.0", features = ["rt", "net", "io-util"] }
log = "0.4.16"
futures = "0.3.21"
num-traits = "0.2.14"
num-derive = "0.3.3"
rusb = "0.9.1"

[dev-dependencies]
tokio = { version = "1.17.0", features = ["full"] }
env_logger = "0.8.3"
```

While your output says you are using tokio 1.12.0. That could be a dependency of your code.

jiegec, 2022-08-11:
Please upgrade your rustc version.

closed 2022-08-11 by jiegec