jiegec/usbip #6: USBIP attached device gets disconnected after a few seconds

joshurawr, 2022-09-06:
I'm trying to pass a USB device from macOS to Linux. It's disconnecting after about 5 seconds. On the macOS host/server, I run `env RUST_LOG=info cargo run --example host`

The client is running Debian 11 and usbip:
Package: usbip
Version: 2.0+5.10.136-1
Priority: optional
Section: admin
Source: linux (5.10.136-1)
Maintainer: Debian Kernel Team <debian-kernel@lists.debian.org>

On the client, I run `sudo usbip list -r 192.168.4.185` (this is the local IP address of the mac), and I get the full list of USB devices exposed by usbip server implementation.

At this point, I get some info logs on the server that seem to be pointing to some issue, although the operation on the client did complete successfully:

`[2022-09-06T22:45:56Z INFO  usbip] Got connection from Ok(192.168.5.245:37120)
[2022-09-06T22:45:56Z INFO  usbip] Handler ended with Err(Custom { kind: UnexpectedEof, error: "early eof" })`

Then I attach a device. In my case:

`sudo usbip attach -r 192.168.4.185 -b 3-16-2`

At this point, I get some success logs from the server, and the client shows the attached device if I run `sudo usbip port`:

Server:
`[2022-09-06T22:49:34Z INFO  usbip] Got connection from Ok(192.168.5.245:58240)
[2022-09-06T22:49:34Z INFO  usbip] Found device "/sys/bus/3/16/2"`

After about 5 seconds, the server throws the same error message from earlier:
`[2022-09-06T22:49:44Z INFO  usbip] Handler ended with Err(Custom { kind: UnexpectedEof, error: "early eof" })`

At this point, the device is no longer attached to the client.

joshurawr, 2022-09-06:
Following up, I found a different behavior with different devices:

     3-16-2: Sony Corp. : DualSense wireless controller (PS5) (054c:0ce6)
           : /sys/bus/3/16/2
           : (Defined at Interface level) (00/00/00)
           :  0 - Audio / Control Device / unknown protocol (01/01/00)
           :  1 - Audio / Streaming / unknown protocol (01/02/00)
           :  2 - Audio / Streaming / unknown protocol (01/02/00)
           :  3 - Human Interface Device / No Subclass / None (03/00/00)

This is the "problem device." I don't know if you might have any intuition about macOS doing something goofy with game controllers that's interfering with the connection or if there might be workarounds.

I did a passthrough on this device and it is able to maintain the attached USB device without the server throwing any errors (it'll stay connected even if I also attach the game controller, when the game controller gets disconnected)

      3-9-4: Blue Microphones : Yeti Stereo Microphone (b58e:9e84)
           : /sys/bus/3/9/4
           : (Defined at Interface level) (00/00/00)
           :  0 - Audio / Control Device / unknown protocol (01/01/00)
           :  1 - Audio / Streaming / unknown protocol (01/02/00)
           :  2 - Audio / Streaming / unknown protocol (01/02/00)

jiegec, 2022-09-08:
I think the error `UnexpectedEof` is due to TCP connection closing.

joshurawr, 2022-09-10:
is there something I can do to resolve it, or get additional logs?

jiegec, 2022-11-18:
@Aussiedler  You need to setup Rust development environment.

jamesadevine, 2023-02-03:
I suspect this issue is caused by reading too much data (> than the ep max packet size) of the usb ip device.

This commit: https://github.com/jiegec/usbip/pull/8/commits/ee51a4003b95ef54400aa6990ce2b05823f03967 fixed the instability with my device.

jiegec, 2024-08-03:
Please try again, and if the issue persists, you can reopen the issue.

closed 2024-08-03 by jiegec