jiegec/usbip #20: Incorrect Constant Values for Standard Device Request Codes

313ctric, 2023-06-23:
In [consts.rs](https://github.com/jiegec/usbip/blob/8506ede125823dba4f46787c23a9a7f7a2a5d320/src/consts.rs#L70) the codes for `SetInterface` and `SynthFrame` (should be `SynchFrame`) are incorrect.
They are currently `0x11` and `0x12` respectively, but should be decimal 11 and 12 (`0x0B` and `0x0C`).

They are specified in Table 9-4 of the USB 2.0 specification.
[https://www.usb.org/document-library/usb-20-specification](https://www.usb.org/document-library/usb-20-specification) (`usb_20.pdf`, page 251).

I would be happy to create a pull request, but it's such a small fix it would probably be quicker for someone else to just do it.

Thanks

jiegec, 2023-06-24:
I was referring to https://www.beyondlogic.org/usbnutshell/usb6.shtml#StandardDeviceRequests, so it seems that the website is wrong as well.

jamesadevine, 2023-06-25:
I have informed the author of usb in a nutshell!

PR #22 (2023-08-14): Correct Standard Request Codes
This fixes #20 by correcting the value.
I also added in the `SetAddress` code as I can't think of a good reason why it shouldn't be there and fixed the spelling of `SynchFrame`.
closed 2023-08-14 by jiegec