jiegec/usbip #23: WebAssembly + WebUSB support

beriberikix, 2023-08-14:
Hello! I'm curious if you or anyone is considering porting this to WebAssembly so that it can be combined with WebUSB. I have a project idea where I transport USB/IP over WebSockets to share device from the browser to a web server.

jiegec, 2023-08-14:
I believe the core protocol part can be built to WebAssembly, but the rusb/tokio parts may need some porting to allow custom transport.

beriberikix, 2023-08-15:
Ah, makes sense. There is a crate for webusb that uses rusb as a backend, but that probably doesn't help here much. https://docs.rs/webusb/latest/webusb/

beriberikix, 2023-08-25:
It's also worth noting that `libusb`, which is what `rusb` is based on, has an [emscripten-based](https://github.com/libusb/libusb/blob/master/libusb/os/emscripten_webusb.cpp) backend that might be useful for anyone else interested in this :)