jiegec/usbip #36: devices no longer recognized since ecc21544

knecht, 2023-12-25:
For example, the cdc_acm_serial example is no longer working in master since  ecc2154444683f0d33b36a45f5ae58b7ea4dd04d 

Steps to reproduce:
```
$ git checkout ecc2154444683f0d33b36a45f5ae58b7ea4dd04d
$ cargo run --example cdc_acm_serial &
$ sudo usbip attach -r 127.0.0.1 -b 0-0-0
$ lsusb | grep 0000
$ dmesg
[1406626.087302] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[1406626.087306] vhci_hcd vhci_hcd.0: devid(0) speed(3) speed_str(high-speed)
[1406626.087313] vhci_hcd vhci_hcd.0: Device attached
[1406626.304190] usb 7-1: new high-speed USB device number 69 using vhci_hcd
```

The previous commit was working as expected
```
$ git checkout ecc2154444683f0d33b36a45f5ae58b7ea4dd04d~1
$ cargo run --example cdc_acm_serial &
$ sudo usbip attach -r 127.0.0.1 -b 0
$ lsusb | grep 0000
Bus 007 Device 068: ID 0000:0000 Manufacturer Product
$ dmesg
[1406416.086748] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[1406416.086753] vhci_hcd vhci_hcd.0: devid(0) speed(3) speed_str(high-speed)
[1406416.086760] vhci_hcd vhci_hcd.0: Device attached
[1406416.304218] usb 7-1: new high-speed USB device number 68 using vhci_hcd
[1406416.418213] usb 7-1: SetAddress Request (68) to port 0
[1406416.676329] usb 7-1: New USB device found, idVendor=0000, idProduct=0000, bcdDevice= 0.00
[1406416.676334] usb 7-1: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[1406416.676337] usb 7-1: Product: Product
[1406416.676339] usb 7-1: Manufacturer: Manufacturer
[1406416.676340] usb 7-1: SerialNumber: Serial
[1406416.840288] cdc_acm 7-1:1.0: ttyACM0: USB ACM device
```

h7x4, 2023-12-25:
Hello @knecht, thank you for opening this issue. I'll be looking into it very soon.

Do you happen to know whether this is an issue with all devices, or if it only happens to some? Also, would you mind running the example with the following envvar `RUST_LOG=trace`?

knecht, 2023-12-26:
Hello, this happens with all devices
```
$ RUST_LOG=trace cargo run --example cdc_acm_serial
[2023-12-26T08:34:00Z INFO  cdc_acm_serial] Simulate a char input
[2023-12-26T08:34:01Z INFO  cdc_acm_serial] Simulate a char input
[2023-12-26T08:34:02Z INFO  cdc_acm_serial] Simulate a char input
[2023-12-26T08:34:03Z INFO  cdc_acm_serial] Simulate a char input
[2023-12-26T08:34:04Z INFO  usbip] Got connection from Ok(127.0.0.1:49830)
[2023-12-26T08:34:04Z TRACE usbip] Got OP_REQ_IMPORT
[2023-12-26T08:34:04Z TRACE usbip] Sent OP_REP_IMPORT
[2023-12-26T08:34:04Z INFO  cdc_acm_serial] Simulate a char input
[2023-12-26T08:34:05Z INFO  cdc_acm_serial] Simulate a char input
[2023-12-26T08:34:06Z INFO  cdc_acm_serial] Simulate a char input
[2023-12-26T08:34:07Z INFO  cdc_acm_serial] Simulate a char input
[2023-12-26T08:34:08Z INFO  cdc_acm_serial] Simulate a char input
[2023-12-26T08:34:09Z INFO  cdc_acm_serial] Simulate a char input
```

PR #37 (2023-12-26): usbip_protocol: fix parser bug on cmd_submit and ret_submit
This PR fixes a critical error in the usbip protocol parser.

From https://docs.kernel.org/usb/usbip_protocol.html (USBIP_CMD_SUBMIT, USBIP_RET_SUBMIT)

> 0x30 - n - transfer_buffer. **If direction is USBIP_DIR_OUT then n equals transfer_buffer_length; otherwise n equals 0**. For ISO transfers the padding between each ISO packets is not transmitted.

> 0x30 - n - transfer_buffer. **If direction is USBIP_DIR_IN then n equals actual_length; otherwise n equals 0**. For ISO transfers the padding between each ISO packets is not transmitted [... cut]
closed 2023-12-26 by jiegec