jiegec/usbip #46: Can't get the example `host` to work

bczhc, 2024-01-18:
The `host` example fails to share a logitech receiver.

## How I reproduced it:

1. Plug in the receiver on my Raspberry PI
2. Set up the server:

    ```console
    ~/usbip master ❯ export RUST_LOG=info
    ~/usbip master ❯ sudo target/release/examples/host
    ```

3. Set up the client on my computer:

    ```console
    ~ ❯ sudo usbip list -r 192.168.124.103                                                                                                                                                     16:28:58
    [sudo] password for bczhc:
    Exportable USB devices
    ======================
     - 192.168.124.103
          4-1-0: Linux Foundation : 3.0 root hub (1d6b:0003)
               : /sys/bus/4/1/0
               : Hub / Unused / unknown protocol (09/00/03)
               :  0 - Hub / Unused / Full speed (or root) hub (09/00/00)
    
          3-1-0: Linux Foundation : 2.0 root hub (1d6b:0002)
               : /sys/bus/3/1/0
               : Hub / Unused / Single TT (09/00/01)
               :  0 - Hub / Unused / Full speed (or root) hub (09/00/00)
    
          2-1-0: Linux Foundation : 3.0 root hub (1d6b:0003)
               : /sys/bus/2/1/0
               : Hub / Unused / unknown protocol (09/00/03)
               :  0 - Hub / Unused / Full speed (or root) hub (09/00/00)
    
         1-47-4: Logitech, Inc. : Unifying Receiver (046d:c52f)
               : /sys/bus/1/47/4
               : (Defined at Interface level) (00/00/00)
               :  0 - Human Interface Device / Boot Interface Subclass / Mouse (03/01/02)
               :  1 - Human Interface Device / No Subclass / None (03/00/00)
    
          1-2-1: VIA Labs, Inc. : Hub (2109:3431)
               : /sys/bus/1/2/1
               : Hub / Unused / Single TT (09/00/01)
               :  0 - Hub / Unused / Full speed (or root) hub (09/00/00)
    
          1-1-0: Linux Foundation : 2.0 root hub (1d6b:0002)
               : /sys/bus/1/1/0
               : Hub / Unused / Single TT (09/00/01)
               :  0 - Hub / Unused / Full speed (or root) hub (09/00/00)
    
    ~ ❯ sudo usbip attach -r 192.168.124.103 -b 1-47-4                                                                                                                                         16:29:06
    ```

## Actual behavior

The cursor won't move when I move the mouse.

## Expected behavior

Not so.

## Additional info

With the original [`usbip`](https://usbip.sourceforge.net) (as the server), this works.

**dmesg**:
```
[114533.584038] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[114533.584049] vhci_hcd vhci_hcd.0: devid(65540) speed(2) speed_str(full-speed)
[... 132 lines cut ...]
[114534.034225] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034227] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034230] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034233] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034235] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034238] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034240] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034243] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034246] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034248] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034251] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034253] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034256] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034259] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034261] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034264] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034266] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034269] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034272] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034274] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034277] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034279] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034282] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
[114534.034285] logitech-djreceiver 0003:046D:C52F.003D: unknown main item tag 0x0
```

yunsmall, 2025-05-12:
It has similiar problems in Android termux. I can list usb devices in my Android phone from client, but when I attach a specific device, client says succesfully attached, while mouse still controls the cursor on my Android phone instead of the client machine.

jiegec, 2025-05-13:
This is a known issue, that complex devices do not work due to unknown reasons. Help is appreciated.

closed 2025-07-20 by github-actions[bot]