jiegec/usbip #47: Debugging Entity not found, ignoring device

brandonros, 2024-01-20:
```
Brandon@AsusRogStrix MINGW64 ~/Desktop/usbip (master)
$ cargo run --example host
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
     Running `target\debug\examples\host.exe`
[2024-01-20T20:41:55Z WARN  usbip] Impossible to share Bus 004 Device 000: ID 8086:7a60: Operation not supported or unimplemented on this platform, ignoring device
[2024-01-20T20:41:55Z WARN  usbip] Impossible to share Bus 004 Device 004: ID 8087:0033: Operation not supported or unimplemented on this platform, ignoring device
[2024-01-20T20:41:55Z WARN  usbip] Impossible to share Bus 001 Device 000: ID 1d6b:0004: Operation not supported or unimplemented on this platform, ignoring device
[2024-01-20T20:41:55Z WARN  usbip] Impossible to share Bus 004 Device 003: ID 322e:2122: Entity not found, ignoring device
[2024-01-20T20:41:55Z WARN  usbip] Impossible to share Bus 004 Device 021: ID 18e1:01b3: Entity not found, ignoring device
[2024-01-20T20:41:55Z WARN  usbip] Impossible to share Bus 002 Device 000: ID 8086:1135: Operation not supported or unimplemented on this platform, ignoring device
[2024-01-20T20:41:55Z WARN  usbip] Impossible to share Bus 003 Device 000: ID 1d6b:0000: Operation not supported or unimplemented on this platform, ignoring device
```

Trying to see what the issue is, opening this issue to put my notes

brandonros, 2024-01-20:
[image]

brandonros, 2024-01-20:
[image]
Why would it say Entity not found... hmm... let me add some logs

brandonros, 2024-01-20:
I got it, this only supports `libusb` devices, not devices with like actual drivers...?

beriberikix, 2024-11-01:
Does #54 help?

brandonros, 2024-11-01:
I'm down to try it but I have a question. Is it a bad idea to try and proxy from Windows to Android?

Basically virtual USB, cross-platform

beriberikix, 2024-11-01:
Not sure what you mean - a windows device as a USB server and Android as the client? That seems reasonable.

beriberikix, 2024-11-01:
That should work in theory, but you would need a USBIP Server on Android and a client on Windows. I haven't tested these but something like this [Android app](https://play.google.com/store/apps/details?id=org.cgutman.usbipserverforandroid) and [this client](https://github.com/vadimgrn/usbip-win2) might work.

brandonros, 2024-11-01:
I wrote the Rust + JNI wrapper to run this as the USBIP Server, the problem is

Android wants to speak libusb

Windows wanted to speak not-libusb

That'll never be compatiable, right?

How can the kernel Windows USB driver (even if it boils down to simple libusb-like commands, reading/writing/transferring/ioctling from endpoints, etc.), are they compatible without a conversion layer?

beriberikix, 2024-11-01:
The conversion layer is the USB/IP Protocol. It shouldn't matter how the client and server are implemented, as long as they implement the same spec. The `usbip-win2` uses all native Windows APIs and networking stack, but that shouldn't matter.

brandonros, 2024-11-01:
https://github.com/vadimgrn/usbip-win2/issues/55#issuecomment-1913692407

Some strangeness on this

closed 2025-06-21 by github-actions[bot]