jiegec/usbip #51: Unable to Passthrough Yubikey from macOS

howyay, 2024-07-05:
I was trying to use the `host` example on macOS Sonoma to passthrough my Yubikey to my Linux machine - I was able to get the device attached, and it shows up in `lsusb` on the remote Linux usbip client. 

However, when I try to use it for SSH (FIDO) or log into websites (WebAuthn), they don't seem to think that my Yubikey is present. On the server end, I can see the following log entry endlessly repeating:
```
[2024-07-05T03:11:57Z DEBUG usbip::host] To host device: ep=UsbEndpoint { address: 129, attributes: 3, max_packet_size: 8, interval: 10 } setup=SetupPacket { request_type: 0, request: 0, value: 0, index: 0, length: 0 } req=[]
```
(with minor variations in the parameters)

I'm not sure how to debug it further, please advise, and thank you for your time!

scj643, 2024-11-12:
Weird I'm not even able to get it to attach at all on my Mac

jiegec, 2024-11-13:
Weird, why does it send an empty setup packet?

scj643, 2025-06-14:
Begone bot