jiegec/usbip #52: Panic when running example with release profile

gsingh93, 2024-09-21:
I tested this on macOS and everything seemed to work fine. Now I'm  trying it in a Linux container on ChromeOS, and while the dev profile at least runs, the release profile panics. Output from running the dev profile:
```
$ env RUST_BACKTRACE=1 RUST_LOG=info cargo run --example host
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.03s
     Running `target/debug/examples/host`
[2024-09-21T06:32:41Z WARN  usbip] Impossible to share Bus 002 Device 001: ID 1d6b:0003: No such device (it may have been disconnected), ignoring device
[2024-09-21T06:32:41Z WARN  usbip] Impossible to share Bus 001 Device 001: ID 1d6b:0002: No such device (it may have been disconnected), ignoring device
^C
```
And now the release profile:
```
$ env RUST_BACKTRACE=1 RUST_LOG=info cargo run --release --example host
    Finished `release` profile [optimized] target(s) in 0.03s
     Running `target/release/examples/host`
thread 'main' panicked at /home/gsingh2011/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rusb-0.9.4/src/context.rs:268:28:
Can't init Global usb context, error Other
stack backtrace:
   0: rust_begin_unwind
   1: core::panicking::panic_fmt
   2: std::sys::sync::once::futex::Once::call
   3: rusb::devices
   4: usbip::UsbIpServer::new_from_host
   5: tokio::runtime::park::CachedParkThread::block_on
   6: tokio::runtime::context::runtime::enter_runtime
   7: tokio::runtime::runtime::Runtime::block_on
   8: host::main
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

FWIW, I'm able to run the `rusb` `list_devices` examples in both dev and release profiles and it works fine and lists the devices.

jiegec, 2024-09-21:
This seems to be a problem on rusb's side.

gsingh93, 2024-09-21:
Maybe, but I can't reproduce with rusb's examples, and it seems the issue occurs early on at the creation of a UsbContext so I'm confused as to what's going on.

closed 2025-06-21 by github-actions[bot]