jiegec/usbip #56: Help with Zooz ZST39 LR Z-Wave stick

37Rb, 2025-01-05:
I'm trying to share a Zooz ZST39 LR Z-Wave stick from OSX to Docker Desktop but it doesn't seem to be working. Here is what I've done based on the this guide.

https://docs.docker.com/desktop/features/usbip/

```
% env RUST_LOG=info cargo run --example host
```

Which appears to work fine. Then

```
% docker run --rm -it --privileged --pid=host alpine
# nsenter -t 1 -m
# usbip list -r host.docker.internal                            # Shows the Z-Wave stick as 20-2-4
# usbip attach -r host.docker.internal -d 20-2-4    # Creates a new device /dev/ttyACM0
```

Then I try to run Z-Wave JS with /dev/ttyACM0 but it doesn't work. Z-Wave JS works fine on this device when I run it directly on OSX. So I assume the issue is with my USB/IP setup. Any suggestions? Or hints on how I might debug this?

I think the device is just a serial port so it should work.

jiegec, 2025-01-05:
You can enable verbose logging of usbip host example to see if there are any potential problems in it.

37Rb, 2025-01-05:
Meaning change RUST_LOG to debug? Here is what I see what I do that. Do you see anything that looks bad?

```
ryan@Hou usbip % env RUST_LOG=debug cargo run --example host                 
warning: non-local `impl` definition, `impl` blocks should be written at the same level as their item
  --> src/consts.rs:46:30
   |
46 | #[derive(Copy, Clone, Debug, FromPrimitive)]
   |                              ^------------
   |                              |
   |                              `FromPrimitive` is not local
   |                              move the `impl` block outside of this constant `_IMPL_NUM_FromPrimitive_FOR_EndpointAttributes`
47 | #[cfg_attr(feature = "serde", derive(Serialize, Deserialize))]
48 | pub enum EndpointAttributes {
   |          ------------------ `EndpointAttributes` is not local
   |
   = note: the derive macro `FromPrimitive` defines the non-local `impl`, and may need to be changed
   = note: the derive macro `FromPrimitive` may come from an old version of the `num_derive` crate, try updating your dependency with `cargo update -p num_derive`
   = note: an `impl` is never scoped, even when it is nested inside an item, as it may impact type checking outside of that item, which can be the case if neither the trait or the self type are at the same nesting level as the `impl`
   = note: items in an anonymous const item (`const _: () = { ... }`) are treated as in the same scope as the anonymous const's declaration for the purpose of this lint
   = note: `#[warn(non_local_definitions)]` on by default
   = note: this warning originates in the derive macro `FromPrimitive` (in Nightly builds, run with -Z macro-backtrace for more info)

warning: non-local `impl` definition, `impl` blocks should be written at the same level as their item
  --> src/consts.rs:64:30
   |
64 | #[derive(Copy, Clone, Debug, FromPrimitive)]
   |                              ^------------
   |                              |
   |                              `FromPrimitive` is not local
   |                              move the `impl` block outside of this constant `_IMPL_NUM_FromPrimitive_FOR_StandardRequest`
65 | #[cfg_attr(feature = "serde", derive(Serialize, Deserialize))]
66 | pub enum StandardRequest {
   |          --------------- `StandardRequest` is not local
   |
   = note: the derive macro `FromPrimitive` defines the non-local `impl`, and may need to be changed
   = note: the derive macro `FromPrimitive` may come from an old version of the `num_derive` crate, try updating your dependency with `cargo update -p num_derive`
   = note: an `impl` is never scoped, even when it is nested inside an item, as it may impact type checking outside of that item, which can be the case if neither the trait or the self type are at the same nesting level as the `impl`
   = note: items in an anonymous const item (`const _: () = { ... }`) are treated as in the same scope as the anonymous const's declaration for the purpose of this lint
   = note: this warning originates in the derive macro `FromPrimitive` (in Nightly builds, run with -Z macro-backtrace for more info)

warning: non-local `impl` definition, `impl` blocks should be written at the same level as their item
  --> src/consts.rs:82:30
   |
82 | #[derive(Copy, Clone, Debug, FromPrimitive)]
   |                              ^------------
   |                              |
   |                              `FromPrimitive` is not local
   |                              move the `impl` block outside of this constant `_IMPL_NUM_FromPrimitive_FOR_DescriptorType`
83 | #[cfg_attr(feature = "serde", derive(Serialize, Deserialize))]
84 | pub enum DescriptorType {
   |          -------------- `DescriptorType` is not local
   |
   = note: the derive macro `FromPrimitive` defines the non-local `impl`, and may need to be changed
   = note: the derive macro `FromPrimitive` may come from an old version of the `num_derive` crate, try updating your dependency with `cargo update -p num_derive`
   = note: an `impl` is never scoped, even when it is nested inside an item, as it may impact type checking outside of that item, which can be the case if neither the trait or the self type are at the same nesting level as the `impl`
   = note: items in an anonymous const item (`const _: () = { ... }`) are treated as in the same scope as the anonymous const's declaration for the purpose of this lint
   = note: this warning originates in the derive macro `FromPrimitive` (in Nightly builds, run with -Z macro-backtrace for more info)

warning: non-local `impl` definition, `impl` blocks should be written at the same level as their item
   --> src/hid.rs:164:30
    |
164 | #[derive(Copy, Clone, Debug, FromPrimitive)]
    |                              ^------------
    |                              |
    |                              `FromPrimitive` is not local
    |                              move the `impl` block outside of this constant `_IMPL_NUM_FromPrimitive_FOR_HidDescriptorType`
165 | #[cfg_attr(feature = "serde", derive(Serialize, Deserialize))]
166 | pub enum HidDescriptorType {
    |          ----------------- `HidDescriptorType` is not local
[... 759 lines cut ...]
[2025-01-05T22:26:24Z DEBUG usbip::host] To host device: ep=UsbEndpoint { address: 130, attributes: 2, max_packet_size: 64, interval: 0 } setup=SetupPacket { request_type: 0, request: 0, value: 0, index: 0, length: 0 } req=[]
[... cut ...]
[2025-01-05T22:26:25Z DEBUG usbip::host] To host device: ep=UsbEndpoint { address: 130, attributes: 2, max_packet_size: 64, interval: 0 } setup=SetupPacket { request_type: 0, request: 0, value: 0, index: 0, length: 0 } req=[]
[2025-01-05T22:26:25Z DEBUG usbip::host] To host device: ep=UsbEndpoint { address: 130, attributes: 2, max_packet_size: 64, interval: 0 } setup=SetupPacket { request_type: 0, request: 0, value: 0, index: 0, length: 0 } req=[]
[2025-01-05T22:26:25Z DEBUG usbip::host] To host device: 
```

37Rb, 2025-01-05:
Inside the container Z-Wave JS gives me this error:

```
ZWaveError: Failed to initialize the driver: ZWaveError: Timeout while waiting for an ACK from the controller (ZW0200)
    at Driver.sendMessage (/usr/lib/node_modules/node-red-contrib-zwave-js/node_modules/zwave-js/src/lib/driver/Driver.ts:6059:23)
    at ZWaveController2.queryCapabilities (/usr/lib/node_modules/node-red-contrib-zwave-js/node_modules/zwave-js/src/lib/controller/Controller.ts:1144:37)
    at Driver.initializeControllerAndNodes (/usr/lib/node_modules/node-red-contrib-zwave-js/node_modules/zwave-js/src/lib/driver/Driver.ts:1665:46)
    at Immediate.<anonymous> (/usr/lib/node_modules/node-red-contrib-zwave-js/node_modules/zwave-js/src/lib/driver/Driver.ts:1466:16) (ZW0100)
```

jiegec, 2025-01-06:
Since usb over ip introduces extra latency, it is possible that the latency exceeds the timeout.

37Rb, 2025-01-07:
Ahh that may be it. I tried increasing the timeouts but no luck.

While digging through the Z-Wave JS documentation to find out how to set the timeouts I discovered that it supports ser2net (which I didn't know existed) so I tried that and it worked. I guess that's easier because it's not trying to emulate a local device so timing doesn't matter as much.

Thanks for the suggestions! I'll go ahead and close the issue.

closed 2025-01-07 by 37Rb